nabaztag2018/pynab #348: Internet connection status : different LED colors does not work

nabaztag2018, 2022-06-02:
The mechanism that should change the color of the bottom LED does not work and may confuse the user.

Violet (=good) color is displayed when there's no  connection and Orange (No Internet connection) is displayed when there's w valid connection.

More over, they are some cases where the bottom LED blinks very rapidly (in orange), do not know it this is related.

f-laurens, 2022-06-02:
> The mechanism that should change the color of the bottom LED does not work and may confuse the user.

The network state check mechanism worked nicely when it was integrated (merge of PR #294).
If there is a regression, it must have been induced by a recent change.

Note also that the check is made by `nabd` when it enters its idle loop. So if state is not idle when network state changes, it may take a moment before new state is displayed.

> More over, they are some cases where the bottom LED blinks very rapidly (in orange), do not know it this is related.

Probably unrelated, since network state check only changes pulse color, not the pulse rate.

f-laurens, 2022-06-02:
The introduction of the comitup package (PR #338) may also have some impact here. If comitup is used:
- when comitup is in _hotspot_ mode (initial state), rabbit will have its own IP address (as a Wifi access point), but no internet access, so bottom LED should pulse orange at this stage (no red pulse, unless nabd has a chance to make network check **before** comitup has set up hotspot);
- when comitup is in _connected_ mode (an existing Wifi access point has been joined through comitup), rabbit will have IP address (and internet access) provided through this access point, so bottom LED should go to violet pulse at this stage (unless the Wifi access point itself has no internet access).

closed 2022-06-04 by nabaztag2018

f-laurens, 2022-06-04:
Why is this closed?
Where was the bug?

nabaztag2018, 2022-06-04:
I do not think it's a bug, it's just that it does not react immediately to a loss of connection (on my experience). I think it's totally OK for the user experience.