nabaztag2018/pynab #384: comitup not installed when performing web update

nabaztag2018, 2022-06-12:
Comitup is not installed when performing an upgrade through the web interface.

I guess, user will have to reconfigure the Wifi connection when it will be installed. Any other way to keep the information ?

pguyot, 2022-06-12:
True, but if user performs web update, wifi is already configured :-)

nabaztag2018, 2022-06-12:
True but what if he needs to change its wifi settings ?

f-laurens, 2022-06-12:
> Comitup is not installed when performing an upgrade through the web interface.

Can this really be considered a bug?:
- If user is able to perform web upgrade, Wifi is already properly configured, and comitup is not needed.
- Comitup is not a Pynab component, so it is out of the scope of Pynab manned upgrades.

If user really needs/wants comitup, the clean solution is to wash a new SD card image.

nabaztag2018, 2022-06-12:
Moving it to "enhancement" 

(I'll link to it when user will complain :-))