nabaztag2018/pynab #386: "Communication with Nabd timed out." (Sleeping ?)

nabaztag2018, 2022-06-12:
While performing NFC/RFID operations, nabd do not seem to be responding.

Nothing particular in nad.log, kern.log and daemon.log

Suspect : the NFC sleep app, seems Nabaztag is sleeping ?? but no push on the button or reconfiguring the sleep time seems to wake him up. Restarted nabclockd -> nothing. 

Only a restart of nabd works.

pguyot, 2022-06-12:
Dupe of #385?

nabaztag2018, 2022-06-12:
It's a little different as there's no way to communicate with nabd, whereas when it is in interactive mode you can see its status is inormation system for example.

cyril1929, 2022-06-28:
Je reproduit exactement le même comportement et de façon systématique par un autre biais.
Étape préliminaire :
- Programmer un tag avec l'action : Horloge -> Réveil
Étape pour reproduire :
- Pendant que le lapin dort (je n'ai pas fait le test s'il est déjà réveillé), passer le tag qui le réveille sous son nez.
Comportement et Symptômes:
- Le lapin comprend bien la commande, vu qu'il bouge les oreilles, et que la LED de sa respiration se met à clignoter.
- Mais c'est tout, il ne fait plus rien d'autre, il est comme figé à part la LED de respiration qui clignote.
- Dans l'interface web, on obtient l'erreur suivante " Communication with Nabd timed out. ".
- Pourtant la commande "systemctl status nabd" indique que le service tourne.
- Et aucune erreur dans les logs "/var/log/nabd.log" .

nabaztag2018, 2022-06-29:
Merci @cyril1929 , ça va permettre d'avancer sur ce bug que je n'arrivais pas à bien reproduire.

cyril1929, 2022-07-01:
Bon en fait, j'ai dit n'importe quoi. C'est pas du tout systématique. C'est complétement aléatoire.
Ça m'arrive relativement souvent, mais c'est pas si simple à reproduire en fait.
En tout cas, quand ça arrive, c'est toujours suite à une lecture de tag (avec la nouvelle carte dans mon cas). Je sais pas encore si ça arrive juste sur les tags avec les actions réveil et dormir, ou si ça arrive avec toutes les actions.

nabaztag2018, 2022-07-02:
Ah ok, c'est pour ça qu'on a du mal à corriger ce bug. J'avais passé 30mn a essayer de le reproduire en mode debug sans succès.