nabaztag2018/pynab #394: Ears driver broken on v1.0.0 images

f-laurens, 2022-06-14:
Eras driver is broken (incorrect build missing `tagtagtag-ears.dtbo` file) on v1.0.0 images that use the 5.15 kernel (non-raspbian images).
See https://github.com/nabaztag2018/pynab/runs/6866385019?check_suite_focus=true and https://github.com/pguyot/tagtagtag-ears/issues/10.

f-laurens, 2022-06-15:
It this point, it would be safer to remove the non-Raspbian images from [v1.0.0](https://github.com/nabaztag2018/pynab/releases/tag/v1.0.0), till problem is fixed and the images are regenerated.

nabaztag2018, 2022-06-15:
Yes you're right. We'll correct that later. Thanks !

f-laurens, 2022-06-15:
Note: the dietpi image should also be removed.

f-laurens, 2022-06-19:
It would be nice to fix this ASAP (at least fix https://github.com/pguyot/tagtagtag-ears/issues/10 now, even if new v1.0.0 images  are  not regenerated).

f-laurens, 2022-11-28:
Closing, since https://github.com/pguyot/tagtagtag-ears/issues/10 is now fixed.

closed 2022-11-28 by f-laurens