nabaztag2018/pynab #400: Add the Version for all software in the Upgrade view.

Nicosss, 2022-06-19:
**Version of pynab:** v1.0.0 fresh install

**Version of Nabaztag:**

- [ ] Nabaztag (v1)
- [x] Nabaztag:Tag (v2)

**Version of TagTagTag:**

- [ ] Maker Faire 2018
- [x] Ulule 05/2019

**The problem**

1. Go to http://nabaztag.local
2. Click on Upgrade
3. Scroll down to the first 2 softwares
4. Missing Version

**Expected behavior**

Get the software version as for others as in **Pynab v0.9.1**.

**Screenshot(s)/video(s) showing the problem**

[image]

f-laurens, 2022-06-19:
Probably due to the fact that on images the local Git repositories are created as shallow clones with depth=1.
Therefore when latest version (only version in shallow clone) has no tag, no **Version** information (from `git describe --long` or `git describe --long --tags`) will be found.

f-laurens, 2022-06-19:
A local workaround would be to manually do on repositories where version information is "missing":
`git fetch --unshallow` or `git fetch --depth=X`  (with increasing _X_ starting from 2, till a tag is found).

pguyot, 2022-06-20:
The reason why we do depth=1 is to save space when building the disk image (and yet, it is still quite large).
When fetching data we should be able to unshallow.

pguyot, 2022-06-21:
The issue is two-fold:
- tags were not fetched systematically
- tags for pynab and sound driver are not set properly.

See also #394.

Closing this for now, as pushing v1.0.0 tag now to release branch doesn't really make sense, it would generate an upgrade on users with no benefit (the commit only affects GitHub action code) and because #402 is after v1.0.0 tag they wouldn't get the version anyway.

Let's try to get the tags correct for v1.0.1.

Nicosss, 2022-06-22:
It's a bit confusing for people who don't know which version they are using. But we will see after a future upgrade as it is not blocking.

f-laurens, 2022-07-18:
Note the version number will be missing only if using the v1.0.0 image.
If v1.0.0 is installed by upgrading from v0.9.1, the Pynab version number will appear as: `v0.9.1-119-gd575cec`.

Nicosss, 2022-07-18:
Indeed, this is what I noticed after the 0.9.1 update. After that I decided that it was time to do a new clean installation with 1.0.0.
With the new rabbits coming in, I thought it would be easier to make sure I was as close as possible to the new setup. However, no new NFC card for me.

PR #402 (2022-06-21): Fetch tags (fix #400)
Should fix #400
commit d72b9f44 (2022-06-21): Fetch tags (fix #400) (#402)