nabaztag2018/pynab #408: NFC tag support?

f-laurens, 2022-06-29:
Is it possible to add support for the following types of tag?:

```
Technology: iso14443a_mifare_classic
UID: 3b:31:84:d7
Status: Unsupported PICC
```
( tags apparently used in _Intratone_ door badges)

```
Technology: iso14443b
UID: 27:a8:e2:52
Status: Unsupported PICC
```
( tags apparently used in _Navigo_ transit cards)

pguyot, 2022-06-29:
These tags are recognized and can be handled by applications that would listen to rfid/* events.

The way NFC/RFID is handled is that we write few bytes on tags with the application configuration. So a tag works on another rabbit or accross reformat. This is not true with ifftt because the action URL is too long as it embeds the API key and it does not fit ST25 512 bits RFID tags sold by Violet. Newer type 2 tags are bigger and could handle these URLs.

We definitely do not want to write Navigo cards. We could decode more of what is on them for a specific application.

Mifare Classic cards are encrypted with a proprietary protocol. I got some code working with cheap
blank cards, this probably has to be into st25r driver as the protocol requires to send non-conforming ISO-14443 frames (parity bits and CRC are encrypted, which is a poor design and helped the protocol being cracked). If this feature is implemented, we could write these tags and let users write applications to them.

mehdichaouch, 2022-06-29:
I have an old navigo card from my college period (almost 18 years), why we do not allow, maybe from a secured area (like `For devs`) to write any nfc support?