nabaztag2018/pynab #409: Voice command get the rabbit stuck with pink nose

zogoon, 2022-07-04:
**Version of pynab:** Raspbian GNU/Linux 10 (buster) - Kernel 5.10.103+ #1529 armv6l
**Version of Nabaztag:** Nabaztag:Tag (v2)

**Version of TagTagTag:** Ulule 05/2019

**The problem**
When pushing button on top for voice command/control, the Nabaztag got stuck with pink noze. Ears are responsive, and after a certain amount of time (multiple minutes), it goes back to normal behavior (language is set to French)

1. click on button on top
2. wait for nose to turn pink + ear sound
3. release button and speak voice command (Meteo, air quality)
4. Nothing happens, nose stays pink, no sound. ears are not responsive at start but after some minutes
5. Rabbit behavior goes back to normal (displaying meteo & air quality + moods + etc.)

**Expected behavior**
Rabbit to answer the question (Meteo, air quality)

**Additional information & file uploads**
Add any other context about the problem here.

It seems that the "coach" question works better, but not all the time.
System is working properly, latest update has been applied. Wake up and sleep function is working properly, hour are announced correctly, ears are responsive. It seems that the voice commands are not properly working and put the rabbit into a blocked state for a few minutes.

mrcroupotin, 2022-07-05:
Step 3 is incorrect. You have to keep pushing the button while speaking. 
how long the rabbit stays in step 4?

zogoon, 2022-07-07:
It stays in step 4 more than 5 minutes (no watchdog surveillance or anything like that seems to stop the proc=ess). Sometimes, it even needs a hard reboot (unplugging power source).

gremlin-58, 2022-08-20:
I tried to reproduce this issue; connecting via ssh, command "top" showed nabd consuming 86% cpu and /opt/pynab/sound_alsa_recording.raw file size was zero.