nabaztag2018/pynab #413: 2022 NFC cards versus ASR

f-laurens, 2022-07-11:
The (some?) 2022 NFC cards appear to make ASR virtually unusable, since they interfere with voice recording by generating a huge continuous 'ploc-ploc' on the left microphone/channel, that obfuscates the voice recorded on the right channel.
See: https://www.tagtagtag.fr/forum/showthread.php?tid=432

Would a software workaround be feasible, by disabling the left mic/recording channel?

- either directly in the `tagtagtag-sound` driver
- or in the calls to `alsa` routines that trigger recording for ASR in `nabd`.

pguyot, 2022-07-12:
This probably is related to enabling/disabling of the NFC field. We can simply stop polling for nfc tags when we start recording, and resume afterwards.

f-laurens, 2022-07-12:
If this is the case, this is good news.
A fix implementing this (to be released as part of a v1.0.1?) would be welcome.
It might also be useful to give users some control by allowing them to enable/disable NFC polling from the NFC web page.

commit 85ac3734 (2022-08-28): Disable NFC/RFID polling during ASR (fix #413)
commit 0ceca610 (2022-08-28): Disable NFC/RFID polling during ASR (fix #413) (#426)
closed 2022-08-28 by pguyot