nabaztag2018/pynab #436: nfc nfc0: st25r391x_turn_field_on: Failed to turn oscillator on: -1

hpiedcoq, 2022-11-19:
**Version of pynab:** v0.xx.xx

**Version of Nabaztag:**

- [ ] Nabaztag:Tag (v2)

**Version of TagTagTag:**

- [ ] Ulule 10/2021

**Version of RFID:**

- [ ] Board NFC 2022 from Ulule 10/2021

**The problem**

Upgdraded to 1.0.1 form the web.

dmesg on the rabbit is pretty verbose with error : 

```
[ 6753.736096] nfc nfc0: st25r391x_turn_field_on: Failed to turn oscillator on: -1
[ 6759.665083] nfc nfc0: st25r391x_turn_field_on: Failed to turn oscillator on: -1
[ 6770.474210] nfc nfc0: st25r391x_turn_field_on: Time out waiting for interrupt bit (adjust regulators command)
[ 6780.065285] nfc nfc0: st25r391x_turn_field_on: Failed to turn oscillator on: -1
[ 6783.817033] nfc nfc0: st25r391x_turn_field_on: Failed to turn oscillator on: -1
[ 6784.667080] nfc nfc0: st25r391x_turn_field_on: Failed to turn oscillator on: -1

```

f-laurens, 2022-11-19:
See issue https://github.com/pguyot/st25r391x/issues/5 ...

closed 2022-11-20 by hpiedcoq

f-laurens, 2022-11-21:
It might be useful to keep this issue open, as a reminder, till NFC driver issue is fixed.