nabaztag2018/pynab #479: Cannot get HEAD - not a git repository?

Madcat-76310, 2024-06-11:
Hi,

I've made this to try to fix this error : "Cannot get number of commits from upstream. Not connected to the internet?"

Nabaztag@Nabaztag:/var/www/html $ sudo apt install git
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn
The following packages will be upgraded:
  git
1 upgraded, 0 newly installed, 0 to remove and 164 not upgraded.
Need to get 4,187 kB of archives.
After this operation, 51.2 kB of additional disk space will be used.
Get:1 http://mirrors.ircam.fr/pub/raspbian/raspbian buster/main armhf git armhf 1:2.20.1-2+deb10u8 [4,187 kB]
Fetched 4,187 kB in 9s (461 kB/s)                                              
Reading changelogs... Done
(Reading database ... 101395 files and directories currently installed.)
Preparing to unpack .../git_1%3a2.20.1-2+deb10u8_armhf.deb ...
Unpacking git (1:2.20.1-2+deb10u8) over (1:2.20.1-2+deb10u4) ...
Setting up git (1:2.20.1-2+deb10u8) ...

And now, i've this error on full lines : "Cannot get HEAD - not a git repository?"

and this : 

Nabaztag@Nabaztag:~ $ sudo git init
Initialized empty Git repository in /home/Nabaztag/.git/
Nabaztag@Nabaztag:~ $ cd /home/Nabaztag/
Nabaztag@Nabaztag:~ $ ls -laf
.bash_history  .gnupg  .  .git  .profile  .bashrc  ..  .bash_logout
Nabaztag@Nabaztag:~ $ cd .git
Nabaztag@Nabaztag:~/.git $ ls
branches  config  description  HEAD  hooks  info  objects  refs

How to fix this ?
Thanks...

Madcat-76310, 2024-06-11:
May be this help : 

Nabaztag@Nabaztag:~ $ sudo service nabweb status
● nabweb.service - Nabaztag web server
   Loaded: loaded (/lib/systemd/system/nabweb.service; enabled; vendor preset: e
   Active: active (running) since Tue 2024-06-11 10:21:16 CEST; 8min ago
 Main PID: 581 (gunicorn)
    Tasks: 3 (limit: 991)
   CGroup: /system.slice/nabweb.service
           ├─581 /opt/pynab/venv/bin/python3.7 /opt/pynab/venv/bin/gunicorn --ti
           └─725 /opt/pynab/venv/bin/python3.7 /opt/pynab/venv/bin/gunicorn --ti

Jun 11 10:25:19 Nabaztag gunicorn[581]:         git config --global --add safe.d
Jun 11 10:25:20 Nabaztag gunicorn[581]: fatal: detected dubious ownership in rep
Jun 11 10:25:20 Nabaztag gunicorn[581]: To add an exception for this directory, 
Jun 11 10:25:20 Nabaztag gunicorn[581]:         git config --global --add safe.d
Jun 11 10:25:20 Nabaztag gunicorn[581]: fatal: detected dubious ownership in rep
Jun 11 10:25:20 Nabaztag gunicorn[581]: To add an exception for this directory, 
Jun 11 10:25:20 Nabaztag gunicorn[581]:         git config --global --add safe.d
Jun 11 10:25:20 Nabaztag gunicorn[581]: fatal: detected dubious ownership in rep
Jun 11 10:25:20 Nabaztag gunicorn[581]: To add an exception for this directory, 
Jun 11 10:25:20 Nabaztag gunicorn[581]:         git config --global --add safe.d
Nabaztag@Nabaztag:~ $

Madcat-76310, 2024-06-11:
Nabaztag@Nabaztag:~/.git $ cat HEAD
ref: refs/heads/master

f-laurens, 2024-06-11:
Please have a look at discussion https://github.com/nabaztag2018/pynab/discussions/476, which covers the "_Cannot get HEAD - not a git repository?_" issue.

Madcat-76310, 2024-06-11:
I've tried #451 and the problem is still here...

f-laurens, 2024-06-12:
What do you mean by _"tried https://github.com/nabaztag2018/pynab/pull/451"_?
See https://www.tagtagtag.fr/forum/showthread.php?tid=491&pid=2726#pid2726.